dkorunic/iSMC #2: Type error runnign iSMC fans

blujio, 2020-01-17:
Darwin MX43TJJG5L 18.7.0 Darwin Kernel Version 18.7.0: Sun Dec  1 18:59:03 PST 2019; root:xnu-4903.278.19~1/RELEASE_X86_64 x86_64

```iSMC fans```

```
Fans:
ERRO[0000] unable to get SMC key F0Sf: unable to convert to float32 type "ui16", bytes [0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0] to float32 
 DESCRIPTION                KEY   VALUE     TYPE 
                 Fan Count  FNum         2  ui8  
 Fan 1 Current Speed        F0Ac  2438 rpm  flt  
 Fan 1 Minimal Speed        F0Mn  2160 rpm  flt  
 Fan 1 Maximum Speed        F0Mx  5927 rpm  flt  
```

dkorunic, 2020-01-18:
Thanks @blujio , it looks like they decided to encode Fan 1 Safe Speed key as uint16 type instead of the usual flt type. I'll add code to handle this in the next release.

dkorunic, 2020-01-21:
Should be triaged with e821fdc913af938e9cf65261a351fb0636973b16 and ready to test in release [v0.2.1](https://github.com/dkorunic/iSMC/releases/tag/v0.2.1). Please test and report back if possible, @blujio

dkorunic, 2020-01-28:
I'll assume this has been fixed, please reopen an issue if the problem still persists.

closed 2020-01-28 by dkorunic