dkorunic/iSMC #3: License issues

xxxserxxx, 2020-06-08:
I'm giving you a heads-up because I have something importing your project and this bit me recently.

iSMC depends on gosmc, which is incorrectly licensed.  It is licensed under MIT but includes smc.c and smc.h, both of which are GPL (in the header). gosmc not only doesn't include a copy of the GPL (as is required by GPL), but GPL software can not be relicensed under MIT.

https://www.tawesoft.co.uk/kb/article/mit-license-faq
> Can I use GPL-licensed code in my MIT-licensed project?
> 
> No. The project as a whole must conform to the terms of the GPL license and must be distributed under the terms of that license. Therefore such a project as a whole must be distributed as GPL, but can still contain MIT-licensed software.

Therefore, my software (MIT) uses your software (MIT) uses gosmc (MIT) uses GPL and we're *all* in legal violation of the GPL.

I'm going to fix my dependencies; I thought you might want to be aware of the liability in yours as well.

dkorunic, 2020-06-09:
Thanks @xxxserxxx , your analysis is correct. Sadly Gosmc links both mentioned files which are GPLv2, I'll have to modify license to GPL. On the other hand, I'm glad you are or were using iSMC!

dkorunic, 2020-06-09:
Fixed in [v0.3.0](https://github.com/dkorunic/iSMC/releases/tag/v0.3.0) release. Thanks!

closed 2020-06-09 by dkorunic