dkorunic/iSMC #7: Command temp shows conversion error

roele, 2022-04-02:
The command `temp` shows a conversion error.

Hardware Model: MacBook M1 Pro
System Version:  macOS 12.3.1

```
  ❯ ./iSMC temp
Temperature:
ERRO[0000] unable to get SMC key TG0H: unable to convert to float32 type "ioft", bytes [0 0 34 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0] to float32
 DESCRIPTION        KEY   VALUE      TYPE
 Battery TS_MAX     TB0T    34.5 °C  flt
 Battery 1          TB1T    33.9 °C  flt
 Battery 2          TB2T    34.5 °C  flt
 Charger Proximity  TCHP    37.5 °C  flt
```

dkorunic, 2022-04-03:
@roele Can you please see if you get any meaningful results from [latest release](https://github.com/dkorunic/iSMC/releases/tag/v0.4.1)? I've built fat/universal binary, it should work on both arm64 and amd64 architecture.

roele, 2022-04-04:
I do get results without any error but some sensors seem to be missing or faulty on the ARM64 architecture. For example there is no temperatures for CPUs, fan current values are missing or battery counts report 0 values etc.

```shell
❯ ./iSMC all
Temperature:
 DESCRIPTION                     KEY   VALUE      TYPE
 Battery TS_MAX                  TB0T    28.0 °C  flt
 Battery 1                       TB1T    27.6 °C  flt
 Battery 2                       TB2T    28.0 °C  flt
 Charger Proximity               TCHP    31.7 °C  flt
 GPU Heatsink                    TG0H    28.0 °C  ioft
 Drive 0 OOBv3 Absolute Raw A    TH0a    27.6 °C  flt
 Drive 0 OOBv3 Absolute Raw B    TH0b    27.7 °C  flt
 Drive 0 OOBv3 Max               TH0x    27.7 °C  flt
 Speaker Proximity               TS0P    34.1 °C  flt
 Airport Proximity               TW0P    33.3 °C  flt
 Power Supply 1 Alt              Tp0C    33.6 °C  flt
 Powerboard Proximity            Tp0P    41.5 °C  flt
 Power Supply 1                  Tp0P    41.5 °C  flt
 PSU Secondary H/S Diode Cooked  Tp0T    44.1 °C  flt
 Palm Rest                       Ts0P    28.0 °C  flt
 Actuator                        Ts1P    27.0 °C  flt

Fans:
 DESCRIPTION          KEY   VALUE     TYPE
 Fan Count            FNum         2  ui8
 Fan 1 Minimal Speed  F0Mn  1499 rpm  flt
 Fan 1 Maximum Speed  F0Mx  4296 rpm  flt
 Fan 2 Minimal Speed  F1Mn  1499 rpm  flt
 Fan 2 Maximum Speed  F1Mx  4744 rpm  flt

Battery:
 DESCRIPTION      KEY   VALUE     TYPE
 Battery Count    BNum         0
 Battery Info     BSIn         0
 Battery Powered  BATP     false

Power:
 DESCRIPTION   KEY   VALUE     TYPE
 S2 Camera     PCMC     0.0 W  flt
 DC In Total   PDTR     0.0 W  flt
 Heatpipe      PHPC     2.3 W  flt
 PBus          PPBR     6.1 W  flt
 System Total  PSTR     4.9 W  flt

Voltage:
 DESCRIPTION        KEY   VALUE     TYPE
 Mainboard S0 Rail  VD0R     0.0 V  flt
 12V Rail           VP0R    12.7 V  flt

Current:
 DESCRIPTION        KEY   VALUE     TYPE
 Camera S2          ICMC     0.0 A  flt
 Mainboard S0 Rail  ID0R     0.0 A  flt
 Charger BMON       IPBR     0.5 A  flt
```

roele, 2022-04-04:
In case of fan current speed, it seems they show up as soon the fans are actually used (e.g. while transcoding a video).

dkorunic, 2022-04-04:
@roele Fan wise, I don't show SMC keys which are all zero, given that there is quite a lot of always-empty SMC keys. In regards to CPU temperatures, it looks like Apple has made M1 Silicon readings available only through IOKit. I'll see what I can do.

DrPsychick, 2022-04-04:
I can test if that helps. my M1 Air results in this:

```
Temperature:
 DESCRIPTION        KEY   VALUE      TYPE
 Battery TS_MAX     TB0T    16.0 °C  flt
 Battery 1          TB1T    16.0 °C  flt
 Battery 2          TB2T    14.6 °C  flt
 Charger Proximity  TCHP    19.0 °C  flt
 GPU Heatsink       TG0H    16.0 °C  ioft
 Drive 0 OOBv3 Max  TH0x    19.0 °C  flt
 Airport Proximity  TW0P    19.7 °C  flt

Fans:
 DESCRIPTION  KEY   VALUE     TYPE
 Fan Count    FNum         0

Battery:
 DESCRIPTION      KEY   VALUE     TYPE
 Battery Count    BNum         0
 Battery Info     BSIn         0
 Battery Powered  BATP     false

Power:
 DESCRIPTION   KEY   VALUE     TYPE
 DC In Total   PDTR     0.0 W  flt
 Heatpipe      PHPC     0.5 W  flt
 PBus          PPBR     4.8 W  flt
 System Total  PSTR     4.4 W  flt

Voltage:
 DESCRIPTION        KEY   VALUE     TYPE
 Mainboard S0 Rail  VD0R     0.0 V  flt
 12V Rail           VP0R    12.0 V  flt

Current:
 DESCRIPTION        KEY   VALUE     TYPE
 Mainboard S0 Rail  ID0R     0.0 A  flt
 Charger BMON       IPBR     0.4 A  flt
```

dkorunic, 2022-04-05:
M1 silicon support will need two-staged fix:
- adding new SMC keys (in progress, I'm almost done)
- adding IOKit CGO part (needs some additional work, not yet started) due to most M1 temperature sensors being available only through HID sensors access.

First part is easy, but the second part will take me a few weeks given the other things at hand.

dkorunic, 2022-04-06:
Actually, I wrote a POC that seems to be working already, although it's a little bit more challenging than expected.

dkorunic, 2022-04-07:
@roele @DrPsychick I've made a new release that should be able to properly fetch HID sensor values on M1, please test when you can: https://github.com/dkorunic/iSMC/releases/tag/v0.5.0

roele, 2022-04-08:
@dkorunic Output looks a lot better now on M1.

[ismc-output.txt](https://github.com/dkorunic/iSMC/files/8448175/ismc-output.txt)

dkorunic, 2022-04-08:
I think we can close this now, agreed?

closed 2022-04-08 by dkorunic

DrPsychick, 2022-04-08:
@dkorunic you were too fast with your releases :)  Awesome work! I gladly tested 0.5.3:

Battery data seems to be missing. I played with that a while ago and use IOReg to get that info. Not sure if that is of any help, I'm not a C coder by profession.
https://github.com/DrPsychick/homebrew-sensei

```
Temperature:
 DESCRIPTION                KEY   VALUE       TYPE
 Airport 1                  TW0P    22.22 °C  flt
 Battery 1                  TB0T    18.20 °C  flt
 Battery 2                  TB1T    18.20 °C  flt
 Battery 3                  TB2T    16.20 °C  flt
 Drive 0 OOBv3 Max          TH0x    21.80 °C  flt
 GPU Heatsink 1             TG0H    18.00 °C  ioft
 NAND                       TH0x    21.80 °C  flt
 ANE MTR Temp Sensor1               30.00 °C  hid
 GPU MTR Temp Sensor1               30.00 °C  hid
 GPU MTR Temp Sensor4               30.00 °C  hid
 ISP MTR Temp Sensor5               30.00 °C  hid
 NAND CH0 temp                      20.00 °C  hid
 PMGR SOC Die Temp Sensor0          21.09 °C  hid
 PMGR SOC Die Temp Sensor1          21.02 °C  hid
 PMGR SOC Die Temp Sensor2          20.88 °C  hid
 PMU2 TR0Z                          51.85 °C  hid
 PMU2 TR1d                          21.30 °C  hid
 PMU2 TR1l                          27.10 °C  hid
 PMU2 TR2d                          21.80 °C  hid
 PMU2 TR2l                          28.60 °C  hid
 PMU2 TR3b                          26.46 °C  hid
 PMU2 TR3d                          21.30 °C  hid
 PMU2 TR4b                          26.56 °C  hid
 PMU2 TR4d                          22.19 °C  hid
 PMU2 TR5b                          27.85 °C  hid
 PMU2 TR5d                          21.63 °C  hid
 PMU2 TR6b                          27.96 °C  hid
 PMU2 TR7b                          28.17 °C  hid
 PMU2 TR8b                          27.96 °C  hid
 PMU TP3w                           27.53 °C  hid
 PMU tcal                           51.85 °C  hid
 PMU tdev1                          21.86 °C  hid
 PMU tdev2                          21.83 °C  hid
 PMU tdev3                          20.96 °C  hid
 PMU tdev4                          22.22 °C  hid
 PMU tdev5                          22.39 °C  hid
 PMU tdev6                          21.90 °C  hid
 PMU tdev7                          21.36 °C  hid
 PMU tdev8                          20.91 °C  hid
 PMU tdie1                          27.53 °C  hid
 PMU tdie2                          24.74 °C  hid
 PMU tdie4                          25.92 °C  hid
 PMU tdie5                          25.92 °C  hid
 PMU tdie6                          26.24 °C  hid
 PMU tdie7                          26.99 °C  hid
 PMU tdie8                          24.10 °C  hid
 SOC MTR Temp Sensor0               20.58 °C  hid
 SOC MTR Temp Sensor1               20.48 °C  hid
 SOC MTR Temp Sensor2               19.64 °C  hid
 eACC MTR Temp Sensor0              18.05 °C  hid
 eACC MTR Temp Sensor3              16.22 °C  hid
 gas gauge battery                  18.00 °C  hid
 gas gauge battery                  18.20 °C  hid
 gas gauge battery                  16.00 °C  hid
 gas gauge battery                  16.20 °C  hid
 gas gauge battery                  18.20 °C  hid
 gas gauge battery                  18.20 °C  hid
 pACC MTR Temp Sensor2              17.06 °C  hid
 pACC MTR Temp Sensor3              19.11 °C  hid
 pACC MTR Temp Sensor4              21.02 °C  hid
 pACC MTR Temp Sensor5              18.55 °C  hid
 pACC MTR Temp Sensor7              21.53 °C  hid
 pACC MTR Temp Sensor8              23.95 °C  hid
 pACC MTR Temp Sensor9              20.22 °C  hid

Fans:
 DESCRIPTION  KEY   VALUE      TYPE
 Fan Count    FNum          0

Battery:
 DESCRIPTION      KEY   VALUE      TYPE
 Battery Count    BNum          0
 Battery Info     BSIn          0
 Battery Powered  BATP      false

Power:
 DESCRIPTION   KEY   VALUE      TYPE
 Battery       PPBR     5.85 W  flt
 Heatpipe      PHPC     3.81 W  flt
 System Total  PSTR     4.77 W  flt

Voltage:
 DESCRIPTION  KEY   VALUE      TYPE
 12V Rail     VP0R    11.93 V  flt
 DC In        VD0R     0.01 V  flt
 PMU2 VR4b            16.26 V  hid
 PMU2 VR4l            16.19 V  hid
 PMU2 VR5b            16.20 V  hid
 PMU2 VR6b            16.42 V  hid
 PMU2 VR6l             4.29 V  hid
 PMU2 VR8l            16.28 V  hid
 PMU2 VRab            16.15 V  hid
 PMU2 VRbl            16.22 V  hid
 PMU2 VRcb            16.23 V  hid
 PMU2 VRcl            16.21 V  hid
 PMU2 VRdb            16.29 V  hid
 PMU2 VRfl             4.29 V  hid
 PMU2 VRhl             4.29 V  hid
 PMU2 VRil             4.29 V  hid
 PMU2 VRkl            16.28 V  hid
 PMU ldo0             16.17 V  hid
 PMU ldo0             16.32 V  hid
 PMU ldo1             15.71 V  hid
 PMU ldo2             16.17 V  hid
 PMU ldo3             16.28 V  hid
 PMU ldo3             15.20 V  hid
 PMU ldo7             16.20 V  hid
 PMU ldo7             16.47 V  hid
 PMU ldo8             16.16 V  hid
 PMU ldo9             16.16 V  hid
 PMU ldo9             16.36 V  hid
 PMU ldo11            16.17 V  hid
 PMU ldo14            16.31 V  hid

Current:
 DESCRIPTION   KEY   VALUE      TYPE
 Charger BMON  IPBR     0.49 A  flt
 PMU2 IR4b             14.83 A  hid
 PMU2 IR4l             15.49 A  hid
 PMU2 IR8l             15.50 A  hid
 PMU2 IRbl             15.03 A  hid
 PMU2 IRcl             15.65 A  hid
 PMU2 IRkl             14.64 A  hid
 PMU ldo0              16.04 A  hid
 PMU ldo0              15.35 A  hid
 PMU ldo3              14.81 A  hid
 PMU ldo7              15.35 A  hid
 PMU ldo9              14.59 A  hid
 PMU ldo11             15.50 A  hid
```