dkorunic/iSMC #8: Add support for JSON output format

roele, 2022-04-06:
While the current output format (table) is suitable for use by humans a more technical output such as JSON would make it easier to use the tool for other purposes and extract single values for example.

Later even more formats like YAML etc. could be added as well.

dkorunic, 2022-04-08:
Sorry, this will need some refactoring to be able to merge in -- do you want me to do this or you feel up to it?

roele, 2022-04-08:
@dkorunic i will rebase this and get it merge ready again

roele, 2022-04-08:
@dkorunic I rebased the PR incorporating your recent changes. While doing so, i took the liberty to turn the new `--ascii` flag into an output type `ascii`. Give it a spin and let me know what you think.

OT: Merging the changes was quite a struggle, it seems that VS Code does not like CGO and my entire system freezes every couple keystrokes, possibly due to regenerating CGO definitions in the background.

dkorunic, 2022-04-08:
@roele Looks great, I'll merge and release ASAP!
In regards to CGO, it is definitely causing pains for `gopls`, I agree. I tend to see slowdowns even in vim-go, although Goland works great (since it doesn't rely on `gopls`).

roele, 2022-04-12:
@dkorunic Guess this can be closed, right?

closed 2022-04-12 by dkorunic