dkorunic/iSMC #18: json output to return float or int instead of string

tenox7, 2023-05-24:
could you please make `-o json` return values in float or int (for rpm) instead all string?

it's really troubling to parse string in to float from supposed to be better than ascii format...

thank you!

dkorunic, 2023-05-24:
That would require all output to be radically different, having unit in separate field (or column, depending on output type). While it is perfectly doable, it would break compatibility for everyone already using iSMC outputs for whatever reason..

tenox7, 2023-05-24:
sorry I just added this to my original comment as well :)

tenox7, 2023-05-24:
ok so maybe we can have two values, one as string and one as other type?

```
  "PMU vbuck5": {
  "key": "",
  "type": "hid",
  "value": "16.15 V",
  "float": 16.15,
  "unit": "V"
},

this would not break existing schema / clients

tenox7, 2023-05-24:
parsing "28.20 °C" in Go while unmarshalling Json is absolutely hideous, requires 10 lines of code, you need to split it in to fields by space than go through strconv.ParseFloat(slice[0]), error check on each 😭

dkorunic, 2023-06-13:
Done in https://github.com/dkorunic/iSMC/releases/tag/v0.7.0, closing.

closed 2023-06-13 by dkorunic