dkorunic/iSMC #23: Adding support Apple Silicon M2 M3

ESXv, 2024-10-02:
Hello @dkorunic! Thx, iSMC is awesome CLI tool!
Are there plans to add keys for Apple Silicon Macs with m2 m3 chip?

Found some keys [here](https://github.com/exelban/stats/blob/9275c412b86d92286f6a6fda1bf6b71b5512569c/Modules/Sensors/values.swift#L506)

And since there is no explicit key such as "CPU 0 Proximity"(TC0P) for Silicon, perhaps it’s worth calculating the average value?

dkorunic, 2024-10-02:
That's great to hear @ESXv, thanks! I'll take a look and see what I can do.

dkorunic, 2024-10-02:
Can you check release [0.8.1](https://github.com/dkorunic/iSMC/releases/tag/v0.8.1) and see if that works better on M2/M3?

ESXv, 2024-10-03:
Ou, you super fast!:) I'll checking this now, it takes a few hours.

ESXv, 2024-10-03:
Some problems here in the list (missing keys of some CPU Cores)
Checked:
Mac14,2 M2 MacBook Air
Mac14,2 M2 MacBook Air
Mac15,7 M3 Pro MacBook Pro 16
Mac15,6 M3 Pro MacBook Pro 14
Mac14,10 M2 Pro MacBook Pro 16

[mac_m2_m3_temps.txt](https://github.com/user-attachments/files/17246228/mac_m2_m3_temps.txt)

Think need to detect mac model and set the list of available for that model keys.

dkorunic, 2024-10-03:
From what I've checked, we don't have specs on all of these SMC keys. It's not a problem to detect them, it's a problem to know where (and to what) they belong to, with proper descriptions.

ESXv, 2024-10-04:
Oh, If i found some specs i'll post to new request. Thank you for response!

closed 2024-10-04 by ESXv

dkorunic, 2026-11-14:
@ESXv I wanted to reach out and ask for `iSMC guess --gpu --save output.json` with latest iSMC release from your hardware, please to further improve support for your hardware. Also if possible make sure to send `sysctl hw.model`. Thank you in advance!