dkorunic/iSMC #25: Strange 8.3 °C temperature...

hsir, 2025-03-31:
After I ran `ismc temp` on my macbook pro (m4 pro), I got very weird looking temperatures like super cold 8.3 °C or 
super hot 9201.14 °C results, I wonder what are these sensors...

```bash
> $ ismc temp

                       Temperature
 DESCRIPTION                   KEY      VALUE       TYPE
 Actuator                      Ts1P      26.4 °C    flt
 Airflow Left                  TaLP      33.9 °C    flt
 Airflow Right                 TaRF      32.9 °C    flt
 Airport                       TW0P      34.4 °C    flt
 Airport 1                     TW0P      34.4 °C    flt
 Battery 1                     TB1T      28.0 °C    flt
 Battery 2                     TB2T      28.1 °C    flt
 Battery 3                     TB2T      28.1 °C    flt
 CPU Efficiency Core 1         Te05      40.7 °C    flt
 CPU Efficiency Core 2         Tp1t      40.0 °C    flt
 CPU Efficiency Core 4         Te0S      39.9 °C    flt
 CPU Performance Core 1        Tp01      40.0 °C    flt
 CPU Performance Core 2        Tp05      40.0 °C    flt
 CPU Performance Core 3        Tp09      40.0 °C    flt
 CPU Performance Core 4        Tp0D      40.0 °C    flt
 CPU Performance Core 5        Tp0X      40.0 °C    flt
 CPU Performance Core 6        Tp0b      40.0 °C    flt
 CPU Performance Core 7        Tp0f      40.0 °C    flt
 CPU Performance Core 8        Tp0b      40.0 °C    flt
 DCIn Air Flow                 Ta0P       8.3 °C    flt
 Drive 0 OOBv3 Absolute Raw A  TH0a      30.7 °C    flt
 Drive 0 OOBv3 Absolute Raw B  TH0b      30.7 °C    flt
 Drive 0 OOBv3 Max             TH0x      30.7 °C    flt
 GPU 1                         Tg05      41.0 °C    flt
 GPU 2                         Tg0j      34.4 °C    flt
 GPU 3                         Tg0L      34.5 °C    flt
 GPU Heatsink 1                TG0H      28.0 °C    ioft
 NAND                          TH0x      30.7 °C    flt
 NAND CH0 temp                 temp     31.00 °C    hid
 PMU tcal                      tcal     51.82 °C    hid
 PMU tdev1                     tdev1    9201.14 °C  hid
 PMU tdev2                     tdev2    9202.86 °C  hid
 PMU tdev3                     tdev3    9202.86 °C  hid
 PMU tdev4                     tdev4    9202.86 °C  hid
 PMU tdev5                     tdev5    9201.14 °C  hid
 PMU tdev6                     tdev6    9201.14 °C  hid
 PMU tdev7                     tdev7    9202.86 °C  hid
 PMU tdev8                     tdev8    9202.86 °C  hid
 PMU tdie1                     tdie1    32.77 °C    hid
 PMU tdie2                     tdie2    32.29 °C    hid
 PMU tdie3                     tdie3    32.45 °C    hid
 PMU tdie4                     tdie4    32.21 °C    hid
 PMU tdie5                     tdie5    33.01 °C    hid
 PMU tdie6                     tdie6    34.37 °C    hid
 PMU tdie7                     tdie7    33.73 °C    hid
 PMU tdie8                     tdie8    33.97 °C    hid
 PMU tdie9                     tdie9    32.69 °C    hid
 PMU tdie10                    tdie10   32.61 °C    hid
 PMU tdie11                    tdie11   33.17 °C    hid
 PMU tdie12                    tdie12   32.85 °C    hid
 PMU tdie13                    tdie13   33.41 °C    hid
 PMU tdie14                    tdie14   32.85 °C    hid
 Palm Rest                     Ts0P      27.6 °C    flt
 Power Supply 1 Alt            Tp0C      40.0 °C    flt
 Powerboard 1                  Tp0P      40.0 °C    flt
 Speaker Proximity             TS0P      33.1 °C    flt
 TBT Thermal Diode             Te0T      42.8 °C    flt
 gas gauge battery             battery  28.10 °C    hid
```

I suppose `DCIn Air Flow  Ta0P 8.3 °C` is related to power management? How come it becomes so cold? 😨

[... 104 lines cut ...]
 TBT Thermal Diode             Te0T      43.5 °C    flt
 gas gauge battery             battery  28.20 °C    hid

                Voltage
 DESCRIPTION  KEY      VALUE     TYPE
 12V Rail     VP0R       11.8 V  flt
 DC In        VD0R       19.8 V  flt
 PMU vbuck0   vbuck0    14.99 V  hid
 PMU vbuck1   vbuck1    16.21 V  hid
 PMU vbuck2   vbuck2    16.14 V  hid
 PMU vbuck3   vbuck3    16.28 V  hid
 PMU vbuck4   vbuck4    16.20 V  hid
 PMU vbuck7   vbuck7    16.22 V  hid
 PMU vbuck9   vbuck9    16.15 V  hid
 PMU vbuck11  vbuck11   16.25 V  hid
 PMU vbuck13  vbuck13   16.27 V  hid
 PMU vldo1    vldo1     16.47 V  hid
 PMU vldo5    vldo5     16.36 V  hid
 PMU vldo7    vldo7     16.47 V  hid
 PMU vldo9    vldo9     16.36 V  hid
 PMU vldo10   vldo10    16.28 V  hid

```

</details>

dkorunic, 2025-03-31:
Thanks @hsir, I'll check. SMC variable type `flt` is in other cases (as you can see above) well calculated, so at the first glance it just looks like a sensor with invalid readings, but I'll double check documentation and sensor readings on boards I have around.

closed 2025-10-21 by dkorunic

dkorunic, 2026-11-14:
@hsir I wanted to reach out and ask for `iSMC guess --gpu --save output.json` with latest iSMC release from your hardware, please to further improve support for your hardware. Also if possible make sure to send `sysctl hw.model`. Thank you in advance!